adafruit/Adafruit_APDS9960 #2: esp32

jschoch, 2017-09-22:
does this work with the esp32?  I have not been able to get the sensor to respond though i can get it to work on an Uno.

ladyada, 2017-09-25:
@deanm1278 wanna try it?

jschoch, 2017-10-02:
Thanks for taking this on!!!  Glad to see it seems to be a bug in the ESP arduino lib and not the adafruit lib!

jschoch, 2018-01-13:
any workaround for this?

ladyada, 2018-01-13:
nope. you can ask espressif for updates - we're using the standard i2c library and there's no other API really

future-reading people: do not post "me too!"'s, we have no control over the esp32 i2c library :) please post only if you have a fix

thx!

jschoch, 2018-01-13:
perhaps using twowire per the suggestion below helps?  If you think it does I can take a stab at trying to implement it.

https://github.com/espressif/arduino-esp32/issues/977

deanm1278, 2018-05-08:
Hey checking in on this issue because it's old. Has this been solved yet?

closed 2018-07-17 by deanm1278