adafruit/Adafruit_APDS9960 #6: Sensor's loop take control of the main arduino loop ?

sylvainbeo, 2018-01-11:
Hi,

Using the Adafruit_APDS9960, i noticed in the source code the following:

```
uint8_t Adafruit_APDS9960::readGesture(void) 
{
	uint8_t toRead, bytesRead;
	uint8_t buf[256];
	unsigned long t;
	uint8_t gestureReceived;
	while(1){
           ...
```

Why is there a while(1) there ? 

This cause the main arduino loop to be stopped because of this while (or maybe i'm doing something wrong ?).

If i run the gesture sensor example, and add a println in the loop(), the print is displayed in the console until we activate the gesture sensor with the hand. Then, the loop is stuck, and the only events we got are from the while(1) loop.

I don't know if i'm clear (and sorry for my english). I wanted the sensor to be a global switch (on/off) to my program, but apparently, this cannot be done that way.

Thanks.

Sylvain.

ladyada, 2018-01-11:
that is correct, it has to quickly get data to determine the gesture, it is not done 'in the background' - if you wanted you could write an interrupt-based library that reads data every once in a while but it would be a full rewrite of this function and we have no plans to do so!

closed 2018-01-11 by ladyada

sylvainbeo, 2018-01-12:
Thanks for the answer.
I will maybe give a try.