adafruit/Adafruit_APDS9960 #8: IR Pulse width

marekhalmo, 2018-02-26:
I'm making a smart node that integrates display, BME280 (temperature/humidity/preassure), TSOP 32238 (38kHz IR sensor) and the apds9960.

When i activate the geasure recognition or proximity sensor using  
```
apds.enableProximity(true);
apds.enableGesture(true);
```
the APDS sensor starts to send IR pulses. I don't know the exact frequency of the IR pulses but the  TSOP 32238 is picking them up. 

I found in [this datasheet](https://www.google.sk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&cad=rja&uact=8&ved=0ahUKEwijxYKKhMPZAhVBIVAKHVjiB9MQFghDMAI&url=https%3A%2F%2Fwww.optimusdigital.ro%2Findex.php%3Fcontroller%3Dattachment%26id_attachment%3D605&usg=AOvVaw2hPKVYDvYMT9uKvnwbKWCT) that the pusles should be 62.5 kHz which is well above  the range of the TSOP.

In the datasheet there is mentioned that you can influence the signal frequency and in code of this library there is

```
enum
{
	APDS9960_GPULSE_4US		= 0x00,
	APDS9960_GPULSE_8US		= 0x01,
	APDS9960_GPULSE_16US	= 0x02,
	APDS9960_GPULSE_32US	= 0x03,
};
```

and 
boolean Adafruit_APDS9960::begin(uint16_t iTimeMS, apds9960AGain_t aGain, uint8_t addr) 
.
.
.
 
  _gpulse.GPLEN = APDS9960_GPULSE_32US;
  _gpulse.GPULSE = 9; //10 pulses
  this->write8(APDS9960_GPULSE, _gpulse.get());
.
.
.
}

So if i understand correctly the pulse width of 32US is arround 31kHz.

I will change the value manually in the lib and test later today if the signals collide and if it affects the signal fo the IR sensor and let you know.

But could you add a possiblity to change the frequency to APDS9960_GPULSE_4US  (250kHz) or a different value when calling begin?

marekhalmo, 2018-02-26:
Just a comment for those searching,
the GPULSE is mentioned [this adafruit datasheet](https://www.google.sk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwijxYKKhMPZAhVBIVAKHVjiB9MQFggoMAA&url=https%3A%2F%2Fcdn.sparkfun.com%2Fassets%2Flearn_tutorials%2F3%2F2%2F1%2FAvago-APDS-9960-datasheet.pdf&usg=AOvVaw1hcGk-okABVSkW8gbtrAuB)

Changing the pulse width will affect the power required for the gesture sensor. If the frequency is set to be higher and the amount of pulses is kept the same the effective time the internal LED is ON will decrease. 

This will decrease the power required by the sensor as stated in Table 5, page 16 of the data sheet mentioned. I'm not sure how it will affect the accuracy of the gestures.

Increasing the frequency of the pulses can induce harmonics which could affect the sensor even more so a real life test is needed.

ladyada, 2019-03-07:
Hiya - i think this is a documentation issue not a code issue? Could you suggest some recommended text we could add somewhere?

marekhalmo, 2019-03-11:
Hello Ladyada
It's a long time since is entered this issue. If i remember correctly your are not able to set the frequency in the code and that was the request - to add a method for that (or a parameter in the begin method)
I changed the value in the library and tested if the TSOP reacts to a 4uS signal - and it does (given the short distance between those). I had to disable the IR receiver (TSOP) during geasure recognition.