adafruit/Adafruit_APDS9960 #11: I have problems with RGB reader value and with de calculate LUX and Color temperature.

sebadc, 2018-08-03:
Hi Guys. I am working whit an APDS-9960, I am try obtain the LUX and Color temperature with Adafruit library, but the result that I obtein is not real. If I read with a luxmeter the result is very diferent, Do you have any Examples with tested code?

 My test code is:
 #include "Adafruit_APDS9960.h"
 Adafruit_APDS9960 apds;
 
 void setup() {
 Serial.begin(115200);
 
 if(!apds.begin()){
 Serial.println("failed to initialize device! Please check your wiring.");
 }
 else Serial.println("Device initialized!");
 //enable color sensign mode
 apds.enableColor(true);
 }
 
 void loop() {
 //create some variables to store the color data in
 uint16_t r, g, b, c;
 //wait for color data to be ready
 while(!apds.colorDataReady()){
 delay(5);
 }
 //get the data and print the different channels
 apds.getColorData(&r, &g, &b, &c);
 Serial.print("red: ");
 Serial.print(r);
 Serial.print(" green: ");
 Serial.print(g);
 Serial.print(" blue: ");
 Serial.print(b);
 Serial.print(" clear: ");
 Serial.println(c);
 Serial.println();
 float luz=apds.calculateLux(r,g,b);
 float color=apds.calculateColorTemperature(r,g,b);
 Serial.print (" LUXES: ");
 Serial.println (luz);
 Serial.print (" Color: ");
 Serial.println (color);
 delay(500);
 }
 Regards. Seba

Rvice, 2018-08-13:
Looking at the equation, there are a few things missing.

1. The Gain is not used in the equation for Lux, it likely should be (maybe just divide the computation by the gain?)
1. Overflow on any channel (visual, red, green and blue) are not checked for. This is a function of aTime (not values for 219 and 246 for atime do not reflect the equation 1025 * cycles)

Regarding actual measurements, I cannot comment on these.

gschorcht, 2018-10-18:
The ALS integration time is not taken into account too. This will also lead to incorrect values if the application does not scale down the RGB values correctly before calling the ```calculateLux``` or ```calculateColorTemperature``` method. Even with the default value of 10 for the ```iTimeMS``` parameter in the "begin" method, the RGB count values are already integrated over 3 ADC conversions.