adafruit/Adafruit_APDS9960 #13: Proximity and gesture at the same time doesn't work

heidrichduarte, 2018-11-30:
My application needs proximity and gesture detection at the same time.
I tried to combine both exemples, and it just doesn't work.

If I take the proximity threshold interruption off, after the first gesture detection, the proximity valeu begans to return a constant value.

If I hold the proximity threshold in the firmware I got no return from the readings.

This is the adapted firmware:

#include "Adafruit_APDS9960.h"
Adafruit_APDS9960 apds;

#define INT_PIN 3

void setup() {
  Serial.begin(115200);
  
  if(!apds.begin()){
    Serial.println("failed to initialize device! Please check your wiring.");
  }
  else Serial.println("Device initialized!");

  //gesture mode will be entered once proximity mode senses something close
  apds.enableProximity(true);
  apds.enableGesture(true);
  
  //apds.setProximityInterruptThreshold(0,50);
  //apds.enableProximityInterrupt();
  //pinMode(INT_PIN, INPUT_PULLUP);
}

// the loop function runs over and over again forever
void loop() {
  //read a gesture from the device
    uint8_t gesture = apds.readGesture();
    if(gesture == APDS9960_DOWN) Serial.println("^");
    if(gesture == APDS9960_UP) Serial.println("v");
    if(gesture == APDS9960_LEFT) Serial.println("x");
    if(gesture == APDS9960_RIGHT) Serial.println("o");
    Serial.println(apds.readProximity());
    //if (!digitalRead(INT_PIN)){
    //  apds.clearInterrupt();
      //Serial.println(apds.readProximity());
    //} 
    delay (300);
}

kbowerma, 2019-03-05:
I get the same results.   I do not even red gestures unless I cover the sensor within 1 inch for 2 seconds.     Seems like that behavior might be related.

hoffmannjan, 2019-03-07:
sorry, it's not possible to achive, you can only pick one mode
[https://media.discordapp.net/attachments/537054717575168000/553262337340211220/unknown.png](diagram)

closed 2019-03-07 by hoffmannjan

maduhaime, 2020-10-20:
Does Color and Proximity can work at the same time?