adafruit/Adafruit_APDS9960 #25: Can you please provide an example of how to use Wire's setPins?

githufsb, 2020-10-18:
WRT https://github.com/espressif/arduino-esp32/issues/3779

Possible to add an example of how to use new setPins function in ESP32 Wire.h?

Thanks!

githufsb, 2020-10-18:
Further info....tried to use existing example, and I'm getting this error.  Bit of a newb, so sorry if i'm doing something wrong.

[E][Wire.cpp:64] setPins(): can not set pins if begin was already called
```

#include <Arduino.h>
#include <Wire.h>

#include "Adafruit_APDS9960.h"

//#define GEST_I2C_ADDRESS  0x39
//#define GEST_I2C_SDA      5
//#define GEST_I2C_SCL      4

Adafruit_APDS9960 apds;

void gesture_init()
{
  //TwoWire *_wire = &Wire;
  //_wire->setPins(5,4);

  Wire.setPins(5,4);

  if(!apds.begin()){
    DPRINT("failed to initialize Gesture device! Please check your wiring.\n");
  }
  else DPRINT("Gesture Device initialized!\n");
...
```

ladyada, 2020-10-18:
please post to the espressif forums for tech support heelp :)

closed 2020-10-18 by ladyada