adafruit/Adafruit_APDS9960 #29: Gesture Sensing & Code Blocking

epiller, 2021-04-28:
Is there any possibility to make the gesture sensing non-blocking? At the moment, when gestures are enabled and something is in close proximity of the sensor, everything in the program is blocked. 
Assuming there's no blocking functions in the user's code, could a function, say, "apds.update()" that's running in the loop function achieve the same functionality without block everything else?
Is this a sensor hardware/firmware limitation or just the way the library was written?

Brok0lis, 2021-05-14:
@epiller There is a while loop in readGesture(). I think that is the issue. I also find the blocking part annoying, but it is also present in the SparkFun library. What I am considering is trying to rewrite it somehow, but I would like to think that it is not as easy as I suspect, since apparently 2 sets of engineers have written a blocking implementation. 

The weird thing to me is that ADC conversion for 2 pairs takes 1.3ms according to the datasheet, and FIFO can hold 32 datasets, so why would it be written blocking? Even with 16Mhz on something like an arduino you should have plenty of time to read it every now and then. If I make something sensible I will leave a message here.

ladyada, 2021-05-16:
if someone can do a PR where the engine uses the FIFO that would be great!

epiller, 2021-05-16:
@Brok0lis Would be pretty awesome if you could pull that off, especially since the same changes could be made on the SparkFun library as well! Keep us posted